bluerobotics/cockpit #1871: Custom data-lake variables do not connect with Input Widgets

ES-Alexander, 2025-05-06:
### Bug description

#1752 introduced persistent custom variables in the data-lake. Connecting a numerical one to a joystick input or compound variable works fine, but when connecting a numerical or boolean one to an Input Widget there is no actual link to the underlying variable - changes to the variable are not reflected in the widget, and changes in the widget (while linked to other Input widgets with the same connected variable) are not reflected in the data lake.

### Steps to reproduce

1. Create a numerical custom variable in the data lake
2. Create a new Input Widget (e.g. toggle-switch)
3. Link it to the new variable
4. Try toggling it, nothing changes in the data-lake display

### Primary pain point(s)

The data-lake is supposed to be a unifying source of data, but that only works if the things it contains can be consistently connected to its input and output sources. Having a special variable type that doesn't connect to Input Widgets makes both systems seem less reliable/useful.

### Additional context

Discovered [while documenting](https://github.com/bluerobotics/Cockpit-docs/pull/29).

### Prerequisites

- [x] I have checked to make sure that a similar request has not already been filed or fixed.

rafaellehmkuhl, 2025-05-06:
Yup. This input-widgets implementation is behind on a lot of stuff. We should give it some care soon.

ES-Alexander, 2025-11-19:
Not sure what changed, but @rafaellehmkuhl noticed something similar to this was happening in a test, and I've just confirmed I'm experiencing it in `v1.17.0-beta.9`, so reopening.

closed 2025-11-21 by ES-Alexander