bluerobotics/cockpit #1989: Discovery vehicle is showing duplicated icons

patrickelectric, 2025-07-16:
### Bug description

[image]

### Steps to reproduce

1. Open cockpit
2. Click to discover vehicles in the network

### Primary pain point(s)

No terrible, just weird

### Prerequisites

- [x] I have checked to make sure that a similar request has not already been filed or fixed.

joaoantoniocardoso, 2025-07-16:
by icons you mean.. entries (vehicles/ips)?

patrickelectric, 2025-07-16:
> by icons you mean.. entries (vehicles/ips)?

Dyslexia

rafaellehmkuhl, 2025-07-16:
Just to be sure, in that case you have 3 vehicles, right? "BlueROV2", "BlueROV" and "Patrick-RPi5".

Showing the same vehicle with 2 different IPs seems fine, but it's strange that it's showing the same IP multiple times.

Thanks for the report. Will investigate.