bluerobotics/cockpit #1994: Critical performance drop after some hours of use

joaoantoniocardoso, 2025-07-18:
Version [1.16.0-beta.7](https://github.com/bluerobotics/cockpit/releases/tag/v1.16.0-beta.7)
Released: 2025-07-02

---

After some hours:

[image]

---

Freshly after reloading the tab:

[image]

rafaellehmkuhl, 2025-07-19:
Do you remember how many hours, more or less? So I can try reproducing it.

joaoantoniocardoso, 2025-07-19:
Ahm, I think it was around two or three hours. 

I'm now with it open for one hour, and I can see that it no longer reaches 75Hz (my monitor refresh rate) when I keep moving my mouse over it.

joaoantoniocardoso, 2025-07-19:
Yes, two hours is enough here:

[image]

rafaellehmkuhl, 2025-07-22:
> Like this?
> 
> [image]
> 
> [Heap-20250722T125955.heapprofile.tar.gz](https://github.com/user-attachments/files/21371303/Heap-20250722T125955.heapprofile.tar.gz)

This one also helps, but the main one is in the performance tab.

joaoantoniocardoso, 2025-07-22:
[Trace-20250722T131902.json](https://github.com/user-attachments/files/21371507/Trace-20250722T131902.json)

rafaellehmkuhl, 2025-11-24:
I wonder if this was finally fixed with #1604.