bluerobotics/cockpit #2033: frontend: clarify Video configuration's "Allowed WebRTC remote IP Address" info

rmackay9, 2025-08-06:
### Current behaviour

When setting up a video stream to cockpit, the user must setup the Settings / Video configuration but I think we should improve the "info" description for the "Allowed WebRTC remote IP Addresses" field.

I think the instructions should be more specific about which device's IP addresses should be placed in the field.  The current instructions are copied below but I'm wonder, which component's "interface" (aka IP address) are we talking about?  The RPI?  the user's PC?  the camera?  something else?  Also let's use the most specific word possible and not use multiple words that mean the same thing.  So for example, let's not use both "interface" and "IP address".  if we're looking for an IP address, let's just say "IP address" because it is the more specific term.

`Select the IP addresses to allow connecting to for WebRTC video streaming. For best performance it is recommended to only use the most reliable interfaces - e.g. avoid wireless interfaces if there is a tethered/wired interface available. If no value is specified, all available routes are allowed.`

### Expected or desired behaviour

Instructions that allow the user to more quickly understand what should be placed in the field

### Prerequisites

- [x] I have checked to make sure that a similar request has not already been filed or fixed.

rmackay9, 2025-08-06:
@rafaellehmkuhl,

BTW, perhaps it is a separate issue (e.g. a bug rather than an incorrect comment) but I think the comment is also incorrect when it says, "If no value is specified, all available routes are allowed."  In my testing if no IP addresses are supplied it fails to show the video.  So basically IP addresses must be supplied.  Below is a screen shot of what I'm seeing

[image]

joaoantoniocardoso, 2025-08-06:
Just a technical note: the addresses in this list are used to filter out unwanted routes during the WebRTC negotiation. Those addresses don't translate 1:1 to interfaces -- each interface can have multiple addresses, thus multiple routes.