bluerobotics/cockpit #2047: Mission in the mission planning is disappearing

rafaellehmkuhl, 2025-08-11:
This happened with a BROV2 connected, on standalone, with "auto switch to flight mode" enabled.

The mission is there in the map widget, even if I restart, but is not in the mission planning mode.

Map widget:

[image]

Mission planning:
[image]

vshie, 2026-08-20:
This is what I've been complaining about - you plan a mission, uplad it, it appears in flight mode, but not when you switch back to mission planning - however if you download the mission (yes, the one you _just_) uploaded, takes awhile for big missions, it will appear and can be edited. Deal breaker!

closed 2026-08-20 by ArturoManzoli

rafaellehmkuhl, 2026-08-21:
@ArturoManzoli can you make sure this was fixed with #2829 so we don't close it without a proper fix?

rafaellehmkuhl, 2026-09-10:
still happening — remount still calls `clearMission()` and reloads a BlueOS-synced draft. reopened as #3045.