bluerobotics/cockpit #2087: HTTP Request Actions cannot override User-Agent header

customrovs-cyber, 2025-08-26:
### Bug description

Even if a custom User-Agent header is added in the HTTP Request Action configuration, Cockpit always sends its default Electron/Chrome User-Agent. The custom value is ignored, preventing integration with devices that require a specific User-Agent.

### Steps to reproduce

Example:
1. Open Settings → Actions → New → HTTP Request.
2. Set URL to any local server (e.g., http://127.0.0.1:1880/test).
3. Add header: User-Agent: CustomAgent/1.0.
4. Run the action (▶ button).
5. Inspect the request headers on the server.

Expected: User-Agent header should be CustomAgent/1.0.
Actual: Cockpit always sends its default Electron/Chrome User-Agent, ignoring the custom value.

### Prerequisites

- [x] I have checked to make sure that a similar request has not already been filed or fixed.

rafaellehmkuhl, 2025-09-01:
@customrovs-cyber thanks for reporting this bug! I have indeed confirmed it and I'm going to open a PR fixing it today.

rafaellehmkuhl, 2025-09-01:
Got a solution already. Will open the PR soon. It will be electron/standalone-only thought, as browsers do not allow us to modify the user-agent header (it is overridden).

customrovs-cyber, 2025-09-01:
@rafaellehmkuhl  Thanks, it's great to hear about the fix for query parameters and User-Agent support in Electron (understandable considering the browser limitations). I'll try it out as soon as it's available.

PR #2094 (2025-09-04): actions: http: Allow settings custom user agent
This allow Cockpit to interface with APIs that require specific user-agents to be calling.

[image]

Fix #2087.
To be merged after #2090.