bluerobotics/cockpit #2088: Add joystick button release events to trigger actions

customrovs-cyber, 2025-08-26:
Problem / Limitation:
Currently, joystick buttons in Cockpit only trigger actions on the press event. There is no way to assign a different action on release. This makes it difficult to implement continuous controls such as camera zoom or focus, where a “start” and “stop” command are both needed.

Proposed solution:
Add support for joystick release events, allowing users to configure two actions per button:

On press → send the “start” action (e.g., zoom in, focus near).

On release → send the “stop” action.

Use case / Why it’s useful:

Allows smooth and intuitive control of IP camera lenses (zoom/focus).

Avoids the unnecessary use of an extra joystick button just to send a “stop” command.

Could also be extended to request the current value of an image parameter (e.g., brightness) and display an on-screen message, helping guide the user during adjustments.

Best regards

patrickelectric, 2025-08-26:
Hi @customrovs-cyber, just for curiosity, which camera are you using that has http controls for focus control ?

customrovs-cyber, 2025-08-26:
Hi @patrickelectric 

This is a custom motorized varifocal IP camera I designed for the BlueROV2. I’ve shared progress, videos and details here: https://discuss.bluerobotics.com/t/ip-camera-4k-60fps-plug-and-play-for-br2/16290
(Photos attached at the end )

The camera exposes simple HTTP CGI endpoints not only for lens control (focus near/far/stop, zoom presets, etc.) but also for all image parameters, recording to SD, snapshots, antifog, WDR, brightness, contrast, shutter, etc.

I originally wrote a small Python controller that sends those HTTP requests and mapped them to keys/joystick inputs. I bundle it as a Windows .exe. However, my goal is to make it work directly through Cockpit, so that all camera controls can also be used seamlessly on Mac and Linux.

Right now I can achieve this by routing Cockpit actions through Node-RED (Cockpit → Node-RED → camera), but it would be much cleaner and lighter to call the camera directly from Cockpit.

That’s why I opened the related issues/requests:

User-Agent cannot be customized from an HTTP Request action (some devices check this).

Query strings defined in the action URL don’t seem to be sent in my tests (e.g. /control.cgi?action=…).

Joystick release event support would allow “press = focus near/far, release = focus stop” without consuming an extra button.

Happy to provide exact CGI examples. Thanks

[image]
[image]
[image]

customrovs-cyber, 2025-09-05:
Hi everyone,

First of all, thank you all for the great work on Cockpit!

I’d like to share some progress regarding the points I raised:

HTTP requests: With the recent fixes, everything is working in Cockpit Electron. Query strings are correctly sent, and I confirmed that the real blocker was the query string handling (user-agent turned out not to be strictly necessary, but I still appreciate the effort to add support for it).

Focus “on release”: Although native support for on-release events would be nice, I managed to achieve the same effect using JavaScript Actions + Data Lake variables (press = focus near/far, release = stop). This solves the issue for my use case without needing Node-RED.

Range parameters (e.g. brightness): I also implemented incremental control with JavaScript Actions, and added a small widget to display the current value.

Cockpit Web: Direct requests still face CORS limitations, but I solved this by routing through a small Node-RED proxy running on BlueOS.

I’m also attaching a link with a folder with all the files I used for both Electron and Cockpit in BlueOS, plus some screenshots of Cockpit conifg, the Data Lake variables (since those can’t be exported directly) and the mini_widget.

https://drive.google.com/drive/folders/1DY9W7tuI4DU-4mdwozmygdFMEs9c3CVg

So far, everything is working smoothly, and I’ll continue to add more parameters and widgets step by step. I’ll also post a demo video on the Blue Robotics forum soon.

Thanks again to all the team for your great support and responsiveness!

customrovs-cyber, 2025-09-10:
https://discuss.bluerobotics.com/t/ip-camera-4k-60fps-plug-and-play-for-br2/16290/65?u=andres